tum-vision/tandem #2: Can be used for measuring distance,if can, how long range can detect ?

Mediumcore, 2021-11-24:
Very cool project, thanks for sharing.
And I wander this project can be used for measuring distance,if can, how long range can detect ?

lkskstlr, 2021-11-26:
Hi @Mediumcore,

Thank you for your interest.

**Camera:**
The tracking is photometric and thus a global shutter camera is necessary. Additionally, good geometric calibration is required but we will release a README on how we did the calibration for the realsense D455.

**Distance Measurement:**
This depends on a number of factors. First, this method works for static environments, so the object would have to be static. Second, it requires the object to be somewhat Lambertian, e.g. not specular. This property should hold for many objects, but would not hold for shiny metal or glass. Third, the camera motion needs to be suitable. The best motion is a translation perpendicular to the axis that connects the camera and the object, but other motions work as well. However, a purely rotational motion would not give any information about the distance to the object. In such cases, a mono-to-depth network would be more appropriate.

Kind regards,
Lukas

Mediumcore, 2021-11-27:
Thank you for the awesome work. Thank you for reply~

closed 2021-12-06 by lkskstlr