tum-vision/tandem #3: About the rendered depth of the last keyframe.

chenerg, 2021-12-09:
Very appreciate your great work! I have some confusion about dense tracking with the rendered depth of the last keyframe. 

> During the dense tracking,  a new frame is tracked against the last keyframe n with the rendered depth of n. 

However, when the last keyframe is tracked, you already predicted a dense depth map of the last keyframe with CVA-MVSNet. And the predicted depth of the last keyframe is used for constructing the TSDF volume. Therefore, is there so much difference between the predicted depth or the rendered depth? I'm curious about how much the difference is.

Lurk3r, 2021-12-09:
你好，盛仔已经收到你的来信，我将尽快回复并与你取得联系，非常感谢。

nanyangnan, 2021-12-09:
Hi @chenerg, thanks for your interest in our work.

Since the rendered depth is from the global TSDF grids, it could deliver more globally consistent depth which reduces the tracking drift.
Table 1 of our [paper](https://arxiv.org/pdf/2111.07418.pdf) shows our evaluation on the EuRoC dataset. The column of "DSO+Dense Depth" shows the results of directly using the dense depth map predicted by CVA-MVSNet for tracking. As you can see there, "Ours" which uses the rendered depth achieved better results.

chenerg, 2021-12-10:
Thanks! That's what I missed.

closed 2021-12-10 by chenerg