tum-vision/tandem #5: cmake error

hai8023, 2021-12-27:
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found version "1.67.0") found components: system thread chrono date_time atomic 
-- Found CUDA: /usr/local/cuda-11.1 (found version "11.1") 
-- Caffe2: CUDA detected: 11.1
-- Caffe2: CUDA nvcc is: /usr/local/cuda-11.1/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda-11.1
-- Caffe2: Header version is: 11.1
-- Found CUDNN: /usr/local/cuda-11.1/lib64/libcudnn.so  
-- Found cuDNN: v7.6.5  (include: /usr/local/cuda-11.1/include, library: /usr/local/cuda-11.1/lib64/libcudnn.so)
-- /usr/local/cuda-11.1/lib64/libnvrtc.so shorthash is 3a20f2b6
-- Autodetected CUDA architecture(s):  7.5
-- Added CUDA NVCC flags for: -gencode;arch=compute_75,code=sm_75
-- Found Torch: /home/haimou/文档/libtorch-cxx11-abi-shared-with-deps-1.9.0+cu111/libtorch/lib/libtorch.so  
-- Found OpenCV: /usr/local (found version "4.2.0") 
-- The CUDA compiler identification is NVIDIA 11.1.105
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Check for working CUDA compiler: /usr/local/cuda-11.1/bin/nvcc - skipped
-- Detecting CUDA compile features
-- Detecting CUDA compile features - done
-- Found CUDAToolkit: /usr/local/cuda-11.1/include (found version "11.1.105") 
-- Found Eigen headers in: /usr/local/include/eigen3
-- Release mode (!= not Debug). Adding -O3;-lineinfo to NVCC FLAGS
--- found PANGOLIN, compiling tandem_pangolin library.
--- found OpenCV (/usr/local/include/opencv4), compiling tandem_opencv library.
--- compiling tandem_dataset.
--- not building tandem_demo, since either don't have openCV, Pangolin, or Realsense.
--- not building realsense_calib_recorder, since either don't have openCV, Pangolin, or Realsense.
-- Configuring done
CMake Error at CMakeLists.txt:131 (add_executable):
  Target "tandem_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:131 (add_executable):
  Target "tandem_dataset" links to target "Eigen3::Eigen" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

hai8023, 2021-12-27:
run `cmake .. \
    -D CMAKE_BUILD_TYPE=Release \
    -D CMAKE_PREFIX_PATH=$TANDEM_LIBTORCH_DIR \
    -D CUDNN_LIBRARY=$TANDEM_CUDNN_LIBRARY \
    -D CUDNN_INCLUDE_PATH=$TANDEM_CUDNN_INCLUDE_PATH`

I have eigen on my computer, have you encountered this problem?

closed 2021-12-30 by hai8023

cybaol, 2021-12-30:
Hi bro, do you solve it ?

hai8023, 2022-01-04:
Need to specify the path of eigen's .cmake file
 
In the CMakeLists.txt   @@file
add set(Eigen3_DIR /usr/lib/cmake/eigen3) 
find_package(Eigen3 3.3 REQUIRED NO_MODULE)

@cybaol

lkskstlr, 2022-01-04:
@hai8023 

Thanks for opening the issue and for providing the fix 🚀