tum-vision/tandem #6: Program crashes during initialization

hjxwhy, 2021-12-28:
It's great, thanks for your code.

I try to run the 'tandem_dem' with my realsense435i, but it can not initialize.  I move the camera very slow and only have translation almost, I used the gdb to find out it crash, it is crash in `refusion::tsdfvh::TsdfVolume::IntegrateScanAsync(refusion::RgbdSensor const&, unsigned char*, float*, refusion::float4x4 const&)`,  here is the output of GDB:
···
#0  0x00005555556cb837 in refusion::tsdfvh::TsdfVolume::IntegrateScanAsync(refusion::RgbdSensor const&, unsigned char*, float*, refusion::float4x4 const&) ()
#1  0x00005555556c702f in DrFusion::IntegrateScanAsync(unsigned char*, float*, float const*) ()
#2  0x000055555564ed73 in TandemBackendImpl::CallSequential() (this=this@entry=0x5555972c7180) at /home/hjx/tandem/tandem/src/tandem/tandem_backend.cpp:166
#3  0x000055555564f51e in TandemBackendImpl::Loop() (this=0x5555972c7180) at /home/hjx/tandem/tandem/src/tandem/tandem_backend.cpp:129
#4  0x00007ffff79babcd in  () at /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#5  0x00007fffec4546db in start_thread (arg=0x7ffdfbfff700) at pthread_create.c:463
#6  0x00007fffd9aec71f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

···

lkskstlr, 2022-01-03:
Hi @hjxwhy,

Thank you for the report.

The function the crash appears in is `IntegrateScanAsync` which integrates the depth map into the TSDF volume. Sadly the gdb output doesn't give more specific information, which might be because it has limited insight into what's happening on the GPU. Overall, this feels like a GPU/CUDA problem to me. Maybe some questions:

- What GPU are you using? A RTX 2080 super should be recommended. What is you CUDA and Nvidia driver version?
- Did you change the image resolutions?
- Did you change the TSDF voxel size? It is hard-coded here (https://github.com/tum-vision/tandem/blob/master/tandem/src/FullSystem/FullSystem.cpp#L260) and I will add it as a command line parameter next week. A bigger voxel size reduces the compute requirements but gives a less detailed reconstruction. Already, 0.02 or 0.04 can make a big difference.

Some ideas for debugging:
- Check if the tracking alone works by disabling the TSDF fusion `TSDF_fusion=0 mesh_extraction_freq=0` should work.
- Check if the MVS depth maps displayed as the right image in the lower part of the pangolin window look OK.

I hope this can help
Best
Lukas

lkskstlr, 2022-02-11:
@hjxwhy I will close this issue due to inactivity but please feel free to re-open if the problem isn't solved. Also feel free to post how the problem was finally resolved.

Best
Lukas

closed 2022-02-11 by lkskstlr