tum-vision/tandem #7: tandem_dataset preset=gui got Segmentation fault

mathmax12, 2021-12-29:
bigtrrssss:~/project/newproject/tandem/tandem$ scripts/tracking_euroc.bash

RUNNING --- results/tracking/dense/euroc/V1_01_easy/0
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.[W BinaryOps.cpp:607] Warning: floor_divide is deprecated, and will be removed in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values.
To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor'). (function operator())
[W TensorShape.cpp:2081] Warning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (function operator())
Pass 'Combine redundant instructions' is not initialized.
Verify if there is a pass dependency cycle.
Required Passes:
scripts/tracking_euroc.bash: line 14: 2268508 **Segmentation fault**      (core dumped) build/bin/tandem_dataset preset=gui result_folder=$result_folder files=$scene/images calib=$scene/camera.txt mvsnet_folder=$mvsnet_folder exit_when_done=1 mode=1 > $result_folder/out.txt

mathmax12, 2021-12-29:
I am using a client to access the server to run this. The above issue happnes using windows as client. I just found when I use linux it works.
close this thread.

closed 2021-12-29 by mathmax12

lkskstlr, 2022-01-04:
Hi @mathmax12 

thanks for opening the issue and for providing the fix for future reference 🚀

PatrickSDUST, 2023-08-04:
hi, I met the same problem. Do you means the mstsc in Windows cannot be used while runing this code?