tum-vision/tandem #8: Use preset=gui didn't get mesh

mathmax12, 2021-12-29:
Hi,
I can sucessfully generated the mesh obj on the Euroc dataset.
My problem is when I set preset=gui I can got the some frames showing on the windows but didn't see the meash.
Could you let me know culprit?
Thanks.
[image]

hai8023, 2021-12-30:
@mathmax12 Hello, do you know what is in camera.txtgroundtruth_tum_rgbd.txtposes_gt.txt? I have just touched this part, not very clear

lkskstlr, 2022-01-03:
Hi @mathmax12,

I am not 100% sure what the issue is here or what you are seeing. The screenshot you posted looks like a debugging image from the original DSO code and not like the TANDEM output. This usually appears when pressing control buttons on the left side of the pangolin window. Normally the mesh should just show up in pangolin after some frames needed for initialisation.

On EuRoC are you also using `preset=gui`? Do you mean the `mesh.obj` generated in the results folder or a live mesh display?

If you could send me your full config, I can have look. I am still at my parent's place so I could do an in-depth check next week.

Best
Lukas

mathmax12, 2022-01-03:
@hai8023 I am also new to this area. Sorry for that.

mathmax12, 2022-01-03:
@lkskstlr 
Thank you for helping. No worries! 
_"On EuRoC are you also using preset=gui? Do you mean the mesh.obj generated in the results folder or a live mesh display?"_
The above window is gotten when I run Tandem on EuRoC and use preset=gui. I want to see the live mesh display.
Sorry for the confusion.
This is the script I used to get the above window.
[image]

lkskstlr, 2022-01-04:
Hi @mathmax12,

no worries. I am sorry of the code doesn't work for you as expected :)

I just tried the exact same change and was able to get the GUI with mesh (I had to use X11 forwarding because I am currently in home office, therefore it might look slightly different then the videos we show).

One thing I could imagine is that you accidentally pressed one of the buttons on the left, which will give you further debugging output similar to the image you showed. Maybe you can make sure that this isn't the case. You can also compare to the screenshot. If all this doesn't help you can send me an e-mail to `lukas.koestler` at `tum.de` and I can see if I can help you over zoom.

Best
Lukas

[image]

mathmax12, 2022-01-18:
@lkskstlr 
I found the forwarding x11 on my side doesn't work as expected.
It works well now when displaying the gui on the machine where the program run.
Thanks.

closed 2022-01-18 by mathmax12