tum-vision/tandem #9: make error related to Sophus

autohtc92, 2022-01-06:
Hi, 
I meet some problem related to sophus while building the project,
the log is :
[ 81%] Building CXX object CMakeFiles/tandem.dir/src/dvo-core/src/dense_tracking_config.cpp.o
[ 83%] Building CXX object CMakeFiles/tandem.dir/src/dvo-core/src/util/histogram.cpp.o
[ 84%] Building CXX object CMakeFiles/tandem.dir/src/dvo-core/src/dense_tracking_impl.cpp.o
[ 86%] Building CXX object CMakeFiles/tandem.dir/src/IOWrapper/OpenCV/ImageDisplay_OpenCV.cpp.o
[ 88%] Building CXX object CMakeFiles/tandem.dir/src/IOWrapper/Pangolin/KeyFrameDisplay.cpp.o
[ 89%] Building CXX object CMakeFiles/tandem.dir/src/dvo-core/src/util/id_generator.cpp.o
[ 91%] Building CXX object CMakeFiles/tandem.dir/src/IOWrapper/Pangolin/DrFrameDisplay.cpp.o
[ 93%] Building CXX object CMakeFiles/tandem.dir/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp.o
[ 94%] Building CXX object CMakeFiles/tandem.dir/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp.o
In file included from /home/hutiancheng/project/tandem/tandem/src/util/settings.cpp:27:0:
/usr/local/include/boost/bind.hpp:41:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
 )
 ^
[ 94%] Built target dr_mvsnet_test
/home/hutiancheng/project/tandem/tandem/src/FullSystem/HessianBlocks.cpp: In member function ‘void dso::FrameHessian::setStateZero(const Vec10&)’:
/home/hutiancheng/project/tandem/tandem/src/FullSystem/HessianBlocks.cpp:84:23: error: ‘template<class Scalar_, int Options> class Sophus::SE3’ used without template parameters
   SE3 EepsP = Sophus::SE3::exp(eps);
                       ^~~
/home/hutiancheng/project/tandem/tandem/src/FullSystem/HessianBlocks.cpp:85:23: error: ‘template<class Scalar_, int Options> class Sophus::SE3’ used without template parameters
   SE3 EepsM = Sophus::SE3::exp(-eps);
                       ^~~
/home/hutiancheng/project/tandem/tandem/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp: In member function ‘void dso::IOWrap::PangolinDSOViewer::run()’:
/home/hutiancheng/project/tandem/tandem/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp:280:27: error: ‘glDrawColoredVertices’ is not a member of ‘pangolin’
                 pangolin::glDrawColoredVertices(
                           ^~~~~~~~~~~~~~~~~~~~~
/home/hutiancheng/project/tandem/tandem/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp:280:27: note: suggested alternative: ‘glDrawColouredCube’
                 pangolin::glDrawColoredVertices(
                           ^~~~~~~~~~~~~~~~~~~~~
                           glDrawColouredCube
/home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp: In member function ‘dso::Vec4 dso::FullSystem::trackNewCoarse(dso::FrameHessian*)’:
/home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:452:46: error: ‘Quaterniond’ is not a member of ‘Sophus’
                                  SE3(Sophus::Quaterniond(1, rotDelta, 0, 0),
                                              ^~~~~~~~~~~
/home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:452:46: note: suggested alternative:
In file included from /usr/local/include/eigen3/Eigen/Geometry:42:0,
                 from /usr/local/include/eigen3/Eigen/Dense:6,
                 from /usr/local/include/sophus/rotation_matrix.hpp:7,
                 from /usr/local/include/sophus/so3.hpp:7,
                 from /usr/local/include/sophus/rxso3.hpp:7,
                 from /usr/local/include/sophus/sim3.hpp:7,
                 from /home/hutiancheng/project/tandem/tandem/src/util/NumType.h:28,
                 from /home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.h:29,
                 from /home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:32:
/usr/local/include/eigen3/Eigen/src/Geometry/Quaternion.h:310:28: note:   ‘Eigen::Quaterniond’
 typedef Quaternion<double> Quaterniond;
                            ^~~~~~~~~~~
/home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:455:46: error: ‘Quaterniond’ is not a member of ‘Sophus’
                                  SE3(Sophus::Quaterniond(1, 0, rotDelta, 0),
                                              ^~~~~~~~~~~
/home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:455:46: note: suggested alternative:
In file included from /usr/local/include/eigen3/Eigen/Geometry:42:0,
                 from /usr/local/include/eigen3/Eigen/Dense:6,
                 from /usr/local/include/sophus/rotation_matrix.hpp:7,
                 from /usr/local/include/sophus/so3.hpp:7,
                 from /usr/local/include/sophus/rxso3.hpp:7,
                 from /usr/local/include/sophus/sim3.hpp:7,
                 from /home/hutiancheng/project/tandem/tandem/src/util/NumType.h:28,
                 from /home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.h:29,
                 from /home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:32:
/usr/local/include/eigen3/Eigen/src/Geometry/Quaternion.h:310:28: note:   ‘Eigen::Quaterniond’
 typedef Quaternion<double> Quaterniond;
                            ^~~~~~~~~~~
/home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:458:46: error: ‘Quaterniond’ is not a member of ‘Sophus’
                                  SE3(Sophus::Quaterniond(1, 0, 0, rotDelta),
                                              ^~~~~~~~~~~
/home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:458:46: note: suggested alternative:
In file included from /usr/local/include/eigen3/Eigen/Geometry:42:0,
[... 368 lines cut ...]
                 from /usr/local/include/eigen3/Eigen/Dense:6,
                 from /usr/local/include/sophus/rotation_matrix.hpp:7,
                 from /usr/local/include/sophus/so3.hpp:7,
                 from /usr/local/include/sophus/rxso3.hpp:7,
                 from /usr/local/include/sophus/sim3.hpp:7,
                 from /home/hutiancheng/project/tandem/tandem/src/util/NumType.h:28,
                 from /home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.h:29,
                 from /home/hutiancheng/project/tandem/tandem/src/FullSystem/FullSystem.cpp:32:
/usr/local/include/eigen3/Eigen/src/Geometry/Quaternion.h:310:28: note:   ‘Eigen::Quaterniond’
 typedef Quaternion<double> Quaterniond;
                            ^~~~~~~~~~~
CMakeFiles/tandem.dir/build.make:209: recipe for target 'CMakeFiles/tandem.dir/src/FullSystem/HessianBlocks.cpp.o' failed
make[2]: *** [CMakeFiles/tandem.dir/src/FullSystem/HessianBlocks.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/tandem.dir/build.make:573: recipe for target 'CMakeFiles/tandem.dir/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp.o' failed
make[2]: *** [CMakeFiles/tandem.dir/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp.o] Error 1
CMakeFiles/tandem.dir/build.make:79: recipe for target 'CMakeFiles/tandem.dir/src/FullSystem/FullSystem.cpp.o' failed
make[2]: *** [CMakeFiles/tandem.dir/src/FullSystem/FullSystem.cpp.o] Error 1
CMakeFiles/Makefile2:190: recipe for target 'CMakeFiles/tandem.dir/all' failed
make[1]: *** [CMakeFiles/tandem.dir/all] Error 2
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2

could you give me some device?
Thanks a lot!

lkskstlr, 2022-01-11:
Hi @autohtc92,

I think make is using the wrong Sophus version for building. We have an older version of Sophus in the `thirdparty` folder that should be used (https://github.com/tum-vision/tandem/tree/master/tandem/thirdparty/Sophus).  You build seems to use a local version: `/usr/local/include/sophus/so3.hpp:7`. If you switch to the one in third party it should work.

Best
Lukas

autohtc92, 2022-01-11:
**hi, @lkskstlr 
Thank for your apply,
actually I have tried the version in thirdparty, but got below error:**
[ 15%] Building CUDA object libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o
/home/hutiancheng/project/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker.cpp:28:9: error: ‘Sophus::SE3 {aka Sophus::SE3Group<double>}’ is not a template
 typedef Sophus::SE3<double> SE3;
         ^~~~~~
/home/hutiancheng/project/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker.cpp:28:29: warning: ‘Sophus::SE3’ is deprecated [-Wdeprecated-declarations]
 typedef Sophus::SE3<double> SE3;
                             ^~~
In file included from /home/hutiancheng/project/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker.cpp:6:0:
/usr/local/include/sophus/se3.hpp:35:26: note: declared here
 typedef SE3Group<double> SE3 EIGEN_DEPRECATED;
                          ^~~
libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/build.make:79: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker.cpp.o' failed
make[2]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:298: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/all' failed
make[1]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 16%] Building NVCC intermediate link file libdr/dr_fusion/CMakeFiles/dr-fusion.dir/dr-fusion_intermediate_link.o
[ 18%] Building CXX object libdr/dr_fusion/CMakeFiles/dr-fusion.dir/src/dr_fusion/dr_fusion.cpp.o
[ 20%] Linking CXX static library lib/libdr-fusion.a
[ 20%] Built target dr-fusion
[ 22%] Linking CXX static library lib/libdr-mvsnet.a
[ 22%] Built target dr-mvsnet
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2

**I don't know why.**

charleshamesse, 2022-01-11:
I've fixed this compilation error by editing L28 of cuda_coarse_tracker.cpp: 
replace `typedef Sophus::SE3<double> SE3;` by `typedef Sophus::SE3d SE3;`

autohtc92, 2022-01-12:
@charleshamesse 
Thanks, it worked!

closed 2022-01-12 by autohtc92