tum-vision/tandem #10: links to target "CUDA::cublas" but the target was not   found.

DormouseLi, 2022-01-12:
Hi，
when I run
 ```
cmake .. \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_PREFIX_PATH=$TANDEM_LIBTORCH_DIR \
    -DCUDNN_LIBRARY=$TANDEM_CUDNN_LIBRARY \
    -DCUDNN_INCLUDE_PATH=$TANDEM_CUDNN_INCLUDE_PATH
```
there are some errors:
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- Found CUDA: /usr/local/cuda (found version "11.1") 
-- Caffe2: CUDA detected: 11.1
-- Caffe2: CUDA nvcc is: /usr/local/cuda/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda
-- Caffe2: Header version is: 11.1
-- Found CUDNN: /usr/local/cuda/lib64/libcudnn.so  
-- Found cuDNN: v8.2.1  (include: /usr/local/cuda/include, library: /usr/local/cuda/lib64/libcudnn.so)
-- /usr/local/cuda/lib64/libnvrtc.so shorthash is 3a20f2b6
-- Autodetected CUDA architecture(s):  8.6
-- Added CUDA NVCC flags for: -gencode;arch=compute_86,code=sm_86
-- Found Torch: /home/libtorch-cxx11-abi-shared-with-deps-1.9.0+cu111/libtorch/lib/libtorch.so  
-- Found OpenCV: /usr/local (found version "4.5.4") 
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- The CUDA compiler identification is NVIDIA 11.1.105
-- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc
-- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc -- works
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Found Eigen headers in: /usr/include/eigen3
-- Release mode (!= not Debug). Adding -O3;-lineinfo to NVCC FLAGS
--- found PANGOLIN, compiling tandem_pangolin library.
--- found OpenCV (/usr/local/include/opencv4), compiling tandem_opencv library.
--- compiling tandem_dataset.
-- TANDEM DEMO will be compiled
-- Realsense Calib Recorded Will be compiled
-- Configuring done
CMake Error at CMakeLists.txt:115 (add_library):
  Target "tandem" links to target "CUDA::cublas" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:142 (add_executable):
  Target "tandem_demo" links to target "CUDA::cublas" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at CMakeLists.txt:131 (add_executable):
  Target "tandem_dataset" links to target "CUDA::cublas" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at libdr/cuda_coarse_tracker/CMakeLists.txt:27 (add_library):
  Target "cuda_coarse_tracker" links to target "CUDA::cublas" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CUDA_DEVICE_LINK_EXECUTABLE
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_CUDA_DEVICE_LINK_EXECUTABLE
-- Generating done

how to fix it ? Thanks!

ko440124, 2022-01-18:
Have you ever modified the cmake file? Try to git clone again.

DormouseLi, 2022-01-25:
> Have you ever modified the cmake file? Try to git clone again.

It does work!!!

pankhurivanjani, 2022-01-26:
I am getting this erro 
```
(tandem) vanjani@port-2118:~/tandem/tandem/build$ cmake ..     -DCMAKE_BUILD_TYPE=Release     -DCMAKE_PREFIX_PATH=$TANDEM_LIBTORCH_DIR     -DCUDNN_LIBRARY=$TANDEM_CUDNN_LIBRARY     -DCUDNN_INCLUDE_PATH=$TANDEM_CUDNN_INCLUDE_PATH
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found version "1.65.1") found components: system thread chrono date_time atomic 
-- Found CUDA: /usr/local/cuda (found version "10.1") 
-- Caffe2: CUDA detected: 10.1
-- Caffe2: CUDA nvcc is: /usr/local/cuda/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda
-- Caffe2: Header version is: 10.1
-- Found CUDNN: /usr/lib/x86_64-linux-gnu/libcudnn.so  
-- Found cuDNN: v8.0.5  (include: /home/vanjani/Downloads/cuda/include, library: /usr/lib/x86_64-linux-gnu/libcudnn.so)
-- /usr/local/cuda/lib64/libnvrtc.so shorthash is 5e8a26c9
-- Autodetected CUDA architecture(s):  6.1
-- Added CUDA NVCC flags for: -gencode;arch=compute_61,code=sm_61
-- Found Torch: /home/vanjani/libtorch/lib/libtorch.so  
-- Found OpenCV: /usr/local (found version "3.4.7") 
-- The CUDA compiler identification is NVIDIA 10.1.243
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Check for working CUDA compiler: /usr/local/cuda/bin/nvcc - skipped
-- Detecting CUDA compile features
-- Detecting CUDA compile features - done
-- Found CUDAToolkit: /usr/local/cuda/include (found version "10.1.243") 
-- Found Eigen headers in: /usr/local/include/eigen3
-- Release mode (!= not Debug). Adding -O3;-lineinfo to NVCC FLAGS
--- found PANGOLIN, compiling tandem_pangolin library.
--- found OpenCV (/usr/local/include;/usr/local/include/opencv), compiling tandem_opencv library.
--- compiling tandem_dataset.
--- not building tandem_demo, since either don't have openCV, Pangolin, or Realsense.
--- not building realsense_calib_recorder, since either don't have openCV, Pangolin, or Realsense.
-- Configuring done
CMake Error at CMakeLists.txt:131 (add_executable):
  Target "tandem_dataset" links to target "CUDA::cublas" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:115 (add_library):
  Target "tandem" links to target "CUDA::cublas" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at libdr/cuda_coarse_tracker/CMakeLists.txt:26 (add_library):
  Target "cuda_coarse_tracker" links to target "CUDA::cublas" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
````

lkskstlr, 2022-02-11:
Thank you @ko440124 for the helpful hint. I will close this issue now :)

closed 2022-02-11 by lkskstlr

AreteQin, 2023-08-10:
Hi,

I encountered the same error. Can you pls let me know what have you changed in the CMakeList to solve the problem?