tum-vision/tandem #11: error: no instance of overloaded function "atomicAdd" matches the argument list

mathmax12, 2022-01-17:
Hi ,
I am tring to build the code with the cuda and cudann.
The installation of cuda and cudnn is good and when build the tandem code I got followings:
Any suggestions are appreciated. 

Scanning dependencies of target cuda_coarse_tracker
[  1%] Building CXX object libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker.cpp.o
[  3%] Building CUDA object libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o
/tDisk-1/workspace/bigtree/workspace/slam/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(355): error: no instance of overloaded function "atomicAdd" matches the argument list
            argument types are: (double *, const double)
          detected during instantiation of "void calcGKernel<TPB,Accum>(float, float, float2, float, int, int, const float *, const float *, const float *, const float *, const float *, const float *, const float *, const float *, Accum *) [with TPB=128, Accum=double]" 
(429): here

/tDisk-1/workspace/bigtree/workspace/slam/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(355): error: no instance of overloaded function "atomicAdd" matches the argument list
            argument types are: (double *, const double)
          detected during instantiation of "void calcGKernel<TPB,Accum>(float, float, float2, float, int, int, const float *, const float *, const float *, const float *, const float *, const float *, const float *, const float *, Accum *) [with TPB=4, Accum=double]" 
(445): here

2 errors detected in the compilation of "/tDisk-1/workspace/bigtree/workspace/slam/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu".
libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/build.make:92: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o' failed
make[2]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o] Error 1
CMakeFiles/Makefile2:298: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/all' failed
make[1]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/all] Error 2
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2

ko440124, 2022-01-18:
Try to add the following code at the beginning of 'cuda_coarse_tracker_private.cu'

```
#if !defined(__CUDA_ARCH__) || __CUDA_ARCH__ >= 600
#else
__device__ double atomicAdd(double* address, double val)
{
    unsigned long long int* address_as_ull =
                              (unsigned long long int*)address;
    unsigned long long int old = *address_as_ull, assumed;

    do {
        assumed = old;
        old = atomicCAS(address_as_ull, assumed,
                        __double_as_longlong(val +
                               __longlong_as_double(assumed)));

    // Note: uses integer comparison to avoid hang in case of NaN (since NaN != NaN)
    } while (assumed != old);

    return __longlong_as_double(old);
}
#endif
```
I find this at [https://docs.nvidia.com/cuda/cuda-c-programming-guide/index.html](url). It works in my case,  however, I haven't tried the evaluation yet.

mathmax12, 2022-01-18:
@ko440124 Thanks for that. After recasting the double* input to float*, it works now.

closed 2022-01-18 by mathmax12