tum-vision/tandem #12: RuntimeError: CUDA out of memory

yohannes-taye, 2022-01-22:
I will like to start by saying thank you for the amazing project and paper! After I cloned the repository I followed the instructions to set up the environment as described in the read me file.  Finally, I downloaded the replica dataset and started the training, but before it even starts it exists with the CUDA out of memory error message. I'm using NVIDIA GeForce RTX 2080 Ti. I will put the full error message below. Any help would be appreciated . Thank you !!
```

GPU available: True, used: True
INFO:lightning:GPU available: True, used: True
No environment variable for node rank defined. Set as 0.
WARNING:lightning:No environment variable for node rank defined. Set as 0.
CUDA_VISIBLE_DEVICES: [0]
INFO:lightning:CUDA_VISIBLE_DEVICES: [0]
Epoch 1:   0%|                                                                                                                                                                     | 0/2714 [00:00<?, ?it/s]Traceback (most recent call last):
  File "train.py", line 128, in <module>
    main(*parse_args(parser))
  File "train.py", line 124, in main
    trainer.fit(model)
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/trainer.py", line 859, in fit
    self.single_gpu_train(model)
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/distrib_parts.py", line 503, in single_gpu_train
    self.run_pretrain_routine(model)
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/trainer.py", line 1015, in run_pretrain_routine
    self.train()
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/training_loop.py", line 347, in train
    self.run_training_epoch()
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/training_loop.py", line 419, in run_training_epoch
    _outputs = self.run_training_batch(batch, batch_idx)
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/training_loop.py", line 597, in run_training_batch
    loss, batch_output = optimizer_closure()
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/training_loop.py", line 561, in optimizer_closure
    output_dict = self.training_forward(split_batch, batch_idx, opt_idx, self.hiddens)
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/training_loop.py", line 727, in training_forward
    output = self.model.training_step(*args)
  File "/home/tmc/tandem/cva_mvsnet/models/tandem.py", line 101, in training_step
    outputs = self(batch)
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/tmc/tandem/cva_mvsnet/models/tandem.py", line 43, in forward
    depth_max=batch['depth_max'],
  File "/home/tmc/miniconda3/envs/tandem/lib/python3.7/site-packages/torch/nn/modules/module.py", line 550, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/tmc/tandem/cva_mvsnet/models/cva_mvsnet.py", line 162, in forward
    volume_gate=self.volume_gates[stage] if self.view_aggregation else None,
  File "/home/tmc/tandem/cva_mvsnet/models/module.py", line 1085, in depth_prediction
    half_pixel_centers=half_pixel_centers
  File "/home/tmc/tandem/cva_mvsnet/models/module.py", line 891, in homo_warping
    warped_src_fea[torch.isnan(warped_src_fea)] = 0
RuntimeError: CUDA out of memory. Tried to allocate 2.34 GiB (GPU 0; 10.76 GiB total capacity; 7.43 GiB already allocated; 740.06 MiB free; 8.94 GiB reserved in total by PyTorch)
Epoch 1:   0%|          | 0/2714 [00:04<?, ?it/s]                                    
```

lkskstlr, 2022-02-11:
Dear @yohannes-taye 

thank you for the kind words :)

This means that your GPU memory is not sufficient. You can reduce the memory requirement by lowering the image resolution, model size, batch size, etc..

Best
Lukas

yohannes-taye, 2022-02-12:
> 

Thank you Lukas. 

Yes I did manage to fix the problem by upgrading my  GPU. Sorry for not updating my progress here.

closed 2022-02-12 by yohannes-taye