tum-vision/tandem #14: Error with cnpy and c++11 compiler

pankhurivanjani, 2022-01-28:
Hello!

I am trying to run the code, while building tandem I am getting 2 major errors from cuda_coarse_tracker side.  I have cuda 11.1, and same cudNN and libtorch version as mentioned in readme. 

```
(tandem-exp) vanjani@port-2118:~/tandem/tandem/build$ make -jConsolidate compiler generated dependencies of target dr-mvsnet
[  1%] Building CXX object libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker.cpp.o
[  3%] Building CUDA object libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o
Consolidate compiler generated dependencies of target dr-fusion
[  6%] Built target dr-mvsnet
[ 22%] Built target dr-fusion
Consolidate compiler generated dependencies of target dr_mvsnet_test
[ 25%] Built target dr_mvsnet_test
/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker.cpp:9:18: fatal error: cnpy.h: No such file or directory
compilation terminated.
libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/build.make:75: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker.cpp.o' failed
make[2]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/c++/5/type_traits:35:0,
                 from /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/utils.h:11,
                 from /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h:16,
                 from /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu:4:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/build.make:89: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o' failed
make[2]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o] Error 1
CMakeFiles/Makefile2:274: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/all' failed
make[1]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2
(tandem-exp) vanjani@port-2118:~/tandem/tandem/build$ 
```

Could you suggest anything to solve these?

I am using  SET(CMAKE_CXX_FLAGS "-std=c++0x") in CMakeLists.txt

Thanks,

pankhurivanjani, 2022-01-28:
I also tried using set (CMAKE_CXX_STANDARD 11)  In CMakeLists.txt but the next error log set comes   

 ```
50%] Building CUDA object CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o
/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(18): error: qualified name is not allowed

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(82): error: this declaration has no storage class or type specifier

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(82): error: expected a ";"

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: this declaration has no storage class or type specifier

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: "numeric_cuda" has already been declared in the current scope

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: expected a ";"

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(69): error: name must be a namespace name

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(70): error: name must be a namespace name

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(70): error: expected a ";"

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(88): error: name followed by "::" must be a class or namespace name

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(88): error: expected an identifier

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(88): error: expected a ";"

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: name followed by "::" must be a class or namespace name

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: expected an identifier

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: expected a ";"

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: name followed by "::" must be a class or namespace name

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: expected an identifier

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: expected a ";"

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(93): error: identifier "R" is undefined

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(94): error: identifier "Ki" is undefined

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(96): error: identifier "t" is undefined

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(99): error: name followed by "::" must be a class or namespace name

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(99): error: expected an identifier

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(99): error: expected a ";"

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: name followed by "::" must be a class or namespace name

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: identifier "R" is undefined

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: identifier "Ki" is undefined

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: identifier "RKi" is undefined

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(103): error: name followed by "::" must be a class or namespace name

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(103): error: expected an identifier

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(103): error: expected a ";"

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(104): error: identifier "xy1" is undefined

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(107): error: name followed by "::" must be a class or namespace name

[... 48 lines cut ...]
/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(133): warning: variable "ptT2_s" was declared but never referenced

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(144): warning: variable "pt3_s" was declared but never referenced

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(88): warning: variable "R_s" was declared but never referenced

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): warning: variable "t_s" was declared but never referenced

/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): warning: variable "Ki_s" was declared but never referenced

[... cut ...]
56 errors detected in the compilation of "/home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu".
CMakeFiles/cuda_coarse_tracker.dir/build.make:75: recipe for target 'CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o' failed
make[2]: *** [CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/cuda_coarse_tracker.dir/all' failed
make[1]: *** [CMakeFiles/cuda_coarse_tracker.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

```

pankhurivanjani, 2022-02-01:
This was compiler version error, this code gives problem with g++5/gcc5 version. 

I got hint from here https://github.com/getkeops/keops/issues/102 and wasable to solve this problem  after upgrading to g++7/gcc+7 by increasing the priority as explained here https://stackoverflow.com/questions/7832892/how-to-change-the-default-gcc-compiler-in-ubuntu

I Hope, it helps future users!

closed 2022-02-01 by pankhurivanjani

wtpro, 2022-03-06:
> I also tried using set (CMAKE_CXX_STANDARD 11) In CMakeLists.txt but the next error log set comes
> 
> ```
> 50%] Building CUDA object CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(18): error: qualified name is not allowed
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(82): error: this declaration has no storage class or type specifier
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(82): error: expected a ";"
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: this declaration has no storage class or type specifier
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: "numeric_cuda" has already been declared in the current scope
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: expected a ";"
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(69): error: name must be a namespace name
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(70): error: name must be a namespace name
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(70): error: expected a ";"
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(88): error: name followed by "::" must be a class or namespace name
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(88): error: expected an identifier
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(88): error: expected a ";"
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: name followed by "::" must be a class or namespace name
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: expected an identifier
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: expected a ";"
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: name followed by "::" must be a class or namespace name
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: expected an identifier
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: expected a ";"
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(93): error: identifier "R" is undefined
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(94): error: identifier "Ki" is undefined
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(96): error: identifier "t" is undefined
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(99): error: name followed by "::" must be a class or namespace name
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(99): error: expected an identifier
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(99): error: expected a ";"
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: name followed by "::" must be a class or namespace name
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: identifier "R" is undefined
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: identifier "Ki" is undefined
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: identifier "RKi" is undefined
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(103): error: name followed by "::" must be a class or namespace name
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(103): error: expected an identifier
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(103): error: expected a ";"
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(104): error: identifier "xy1" is undefined
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(107): error: name followed by "::" must be a class or namespace name
> 
[... 51 lines cut ...]
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(88): warning: variable "R_s" was declared but never referenced
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): warning: variable "t_s" was declared but never referenced
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): warning: variable "Ki_s" was declared but never referenced
> 
> /home/vanjani/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(99): warning: variable "RKi_s" was declared but never referenced
> 
[... cut ...]
> CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/cuda_coarse_tracker.dir/all' failed
> make[1]: *** [CMakeFiles/cuda_coarse_tracker.dir/all] Error 2
> Makefile:90: recipe for target 'all' failed
> make: *** [all] Error 2
> ```
> 
> I can see this line in the cmakelists of cudacoarser https://github.com/tum-vision/tandem/blob/master/tandem/libdr/cuda_coarse_tracker/CMakeLists.txt#L23 @lkskstlr Did you also had this problem while developing?

I am having this exact error but I am using gcc-7. Any idea on how to solve this?