tum-vision/tandem #15: not able to open "tandem/cva_mvsnet/model.pt"

PeterFWS, 2022-02-01:
After running the command for the given dataset euroc, the system seems to initialize successfully but is not able to open the "model.pt" using cva_mvsnet.
The flag "**_mvsnet_folder=Folder that contains the exported MVSNet_**." is not very clearly explained. Following is my log from the terminal:

**~/tandem/tandem$** ./build/bin/tandem_dataset preset=dataset result_folder=./results files=/home/shu/Downloads/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/images   calib=/home/shu/Downloads/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/camera.txt mvsnet_folder=/home/shu/Examples/tandem/cva_mvsnet    mode=1

loading data from /home/shu/Downloads/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/images!
loading calibration from /home/shu/Downloads/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/camera.txt!
Loading MVSNet from /home/shu/Examples/tandem/cva_mvsnet/!
PHOTOMETRIC MODE WITHOUT CALIBRATION!

=============== TANDEM Settings: ===============
	Setting 'dataset':
	- no real-time enforcing
	- 2000 active points
	- 5-7 active frames
	- 1-6 LM iteration each KF
	- TSDF fusion: yes
	- dense tracking on cpu (step=1)
	- Pangolin
	  - Fullscreen: 0
	  - Mesh: 1
	  - Smaller Images: 1

Reading Calibration from file /home/shu/Downloads/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/camera.txt ... found!
Input resolution: 640 480
In: 436.225021 436.225021 310.162763 256.946041 0.000000
Out: Rectify Crop
Output resolution: 640 480
finding CROP optimal new model!
initial range: x: -0.7181 - 0.7613; y: -0.5949 - 0.5141!
iteration 00001: range: x: -0.7145 - 0.7575; y: -0.5949 - 0.5141!
iteration 00002: range: x: -0.7109 - 0.7537; y: -0.5949 - 0.5141!
iteration 00003: range: x: -0.7109 - 0.7537; y: -0.5919 - 0.5115!
iteration 00004: range: x: -0.7109 - 0.7537; y: -0.5890 - 0.5090!
iteration 00005: range: x: -0.7109 - 0.7537; y: -0.5890 - 0.5090!

Rectified Kamera Matrix:
 436.27       0 310.165
      0  436.28  256.95
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file 
PhotometricUndistorter: Could not open file!
set timestamps and exposures to zero!
set EXPOSURES to zero!
got 2778 images and 0 timestamps and 0 exposures.!
ImageFolderReader: got 2778 files in /home/shu/Downloads/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/images!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!

----DRMVSNET Initalizing fusion----

----DRMVSNET Initalizing fusion done----
DrMvsnet torch::cuda::is_vailable == true --> seems good
**terminate called after throwing an instance of 'c10::Error'
  what():  open file failed, file path: /home/shu/Examples/tandem/cva_mvsnet/model.pt**
Exception raised from FileAdapter at ../caffe2/serialize/file_adapter.cc:11 (most recent call first):
frame #0: c10::Error::Error(c10::SourceLocation, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) + 0x69 (0x7f8ea507c1d9 in /home/shu/Examples/libtorch/lib/libc10.so)
frame #1: c10::detail::torchCheckFail(char const*, char const*, unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) + 0xd2 (0x7f8ea5078812 in /home/shu/Examples/libtorch/lib/libc10.so)
frame #2: caffe2::serialize::FileAdapter::FileAdapter(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) + 0x1ff (0x7f8ea7ddbc0f in /home/shu/Examples/libtorch/lib/libtorch_cpu.so)
frame #3: torch::jit::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, c10::optional<c10::Device>, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&) + 0x40 (0x7f8ea92e4b50 in /home/shu/Examples/libtorch/lib/libtorch_cpu.so)
frame #4: torch::jit::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, c10::optional<c10::Device>) + 0x6f (0x7f8ea92e4cdf in /home/shu/Examples/libtorch/lib/libtorch_cpu.so)
frame #5: <unknown function> + 0x16b827 (0x55a76f9e3827 in ./build/bin/tandem_dataset)
frame #6: <unknown function> + 0x163482 (0x55a76f9db482 in ./build/bin/tandem_dataset)
frame #7: <unknown function> + 0x3a6bc (0x55a76f8b26bc in ./build/bin/tandem_dataset)
frame #8: <unknown function> + 0x2980d (0x55a76f8a180d in ./build/bin/tandem_dataset)
frame #9: <unknown function> + 0xbd6df (0x7f8e526046df in /usr/lib/x86_64-linux-gnu/libstdc++.so.6)
frame #10: <unknown function> + 0x76db (0x7f8ea55006db in /lib/x86_64-linux-gnu/libpthread.so.0)
frame #11: clone + 0x3f (0x7f8e51cc171f in /lib/x86_64-linux-gnu/libc.so.6)

Aborted (core dumped)

pankhurivanjani, 2022-02-01:
> **~/tandem/tandem$** ./build/bin/tandem_dataset preset=dataset result_folder=./results files=/home/shu/Downloads/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/images [...]

There seems to be some typo with readme so the model has been provided but it is there in this path 

`tandem/tandem/exported/tandem `

This is my output:

```
(tandem-exp) vanjani@port-2118:~/tandem/tandem/bt=dataset result_folder=/home/vanjani/results filees=/home/vanjani/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/images calib=/home/vanjani/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/camera.txt mvsnet_folder=/home/vanjani/tandem/tandem/exported/tandem mode=1
could not parse argument "filees=/home/vanjani/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/images"!!!!
loading calibration from /home/vanjani/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/camera.txt!
Loading MVSNet from /home/vanjani/tandem/tandem/exported/tandem/!
PHOTOMETRIC MODE WITHOUT CALIBRATION!

=============== TANDEM Settings: ===============
	Setting 'dataset':
	- no real-time enforcing
	- 2000 active points
	- 5-7 active frames
	- 1-6 LM iteration each KF
	- TSDF fusion: yes
	- dense tracking on cpu (step=1)
	- Pangolin
	  - Fullscreen: 0
	  - Mesh: 1
	  - Smaller Images: 1

Reading Calibration from file /home/vanjani/euroc_tandem_format_1.1.beta/euroc_tandem_format/V1_01_easy/camera.txt ... found!
Input resolution: 640 480
In: 436.225021 436.225021 310.162763 256.946041 0.000000
Out: Rectify Crop
Output resolution: 640 480
finding CROP optimal new model!
initial range: x: -0.7181 - 0.7613; y: -0.5949 - 0.5141!
iteration 00001: range: x: -0.7145 - 0.7575; y: -0.5949 - 0.5141!
iteration 00002: range: x: -0.7109 - 0.7537; y: -0.5949 - 0.5141!
iteration 00003: range: x: -0.7109 - 0.7537; y: -0.5919 - 0.5115!
iteration 00004: range: x: -0.7109 - 0.7537; y: -0.5890 - 0.5090!
iteration 00005: range: x: -0.7109 - 0.7537; y: -0.5890 - 0.5090!

Rectified Kamera Matrix:
 436.27       0 310.165
      0  436.28  256.95
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file 
PhotometricUndistorter: Could not open file!
got 0 images and 0 timestamps and 0 exposures.!
ImageFolderReader: got 0 files in !
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!

----DRMVSNET Initalizing fusion----

----DRMVSNET Initalizing fusion done----
DrMvsnet torch::cuda::is_vailable == true --> seems good
View Num: 7, ref index: 5
[W BinaryOps.cpp:467] Warning: floor_divide is deprecated, and will be removed in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values.
To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor'). (function operator())
Correctness:
	Depth correct     : 1, error: 0.000665141
	Confidence correct: 1, error: 0.000441842
Correctness:
	Depth correct     : 1, error: 0.000665178
	Confidence correct: 1, error: 0.000441805
Correctness:
	Depth correct     : 1, error: 0.000665188
	Confidence correct: 1, error: 0.000441806
Correctness:
	Depth correct     : 1, error: 0.000665189
	Confidence correct: 1, error: 0.000441808
Correctness:
	Depth correct     : 1, error: 0.000665169
	Confidence correct: 1, error: 0.000441808
Correctness:
	Depth correct     : 1, error: 0.000665172
	Confidence correct: 1, error: 0.000441816
Correctness:
	Depth correct     : 1, error: 0.00066519
	Confidence correct: 1, error: 0.000441807
Correctness:
	Depth correct     : 1, error: 0.000665184
	Confidence correct: 1, error: 0.000441808
Correctness:
	Depth correct     : 1, error: 0.000665186
	Confidence correct: 1, error: 0.000441816
Performance:
	CallAsync     : 8.70475 ms
	Ready         : 0 ms
	GetResult     : 406.197 ms
All looks good!
Segmentation fault (core dumped)
```

lkskstlr, 2022-02-11:
@pankhurivanjani Thank you for the helpful comment :)

@PeterFWS I will close this issue, feel free to re-open if necessary.

closed 2022-02-11 by lkskstlr