tum-vision/tandem #20: No such file or directory

Xie-HIT, 2022-03-10:
When I make the project, I get
```
fatal error: cub/block/block_reduce.cuh: No such file or directory
 #include <cub/block/block_reduce.cuh>
```
but I find this .cuh file in my computer. I don't know how to do next, does anyone have the same question?
My environment: Ubuntu18.04, CUDA10.2

Xie-HIT, 2022-03-10:
I fix that problem by replacing ` #include <cub/block/block_reduce.cuh>` with `#include <thrust/system/cuda/detail/cub/block/block_reduce.cuh>`, since before CUDA11.0, CUB library is included in Thrust(see [https://github.com/NVlabs/cub](url)). But now I am stuck by the following error
`-- Caffe2: CUDA detected: 10.2
-- Caffe2: CUDA nvcc is: /usr/local/cuda-10.2/bin/nvcc
-- Caffe2: CUDA toolkit directory: /usr/local/cuda-10.2
-- Caffe2: Header version is: 10.2
-- Found cuDNN: v7.6.5  (include: /usr/local/cuda-10.2/include, library: /usr/local/cuda-10.2/lib64/libcudnn.so)
-- /usr/local/cuda-10.2/lib64/libnvrtc.so shorthash is 16439d37
-- Automatic GPU detection failed. Building for common architectures.
-- Autodetected CUDA architecture(s): 3.5;5.0;5.2;6.0;6.1;7.0;7.5;7.5+PTX
-- Added CUDA NVCC flags for: -gencode;arch=compute_35,code=sm_35;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_52,code=sm_52;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_75,code=compute_75
-- Unable to find cublas_v2.h in either "/usr/local/cuda-10.2/include" or "/usr/math_libs/include"
-- Found Eigen headers in: /usr/local/include/eigen3
-- Release mode (!= not Debug). Adding -O3;-lineinfo to NVCC FLAGS
--- found PANGOLIN, compiling tandem_pangolin library.
--- found OpenCV (/usr/include;/usr/include/opencv), compiling tandem_opencv library.
--- compiling tandem_dataset.
--- not building tandem_demo, since either don't have openCV, Pangolin, or Realsense.
--- not building realsense_calib_recorder, since either don't have openCV, Pangolin, or Realsense.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xiechen/tandem/tandem/build
Consolidate compiler generated dependencies of target cuda_coarse_tracker
[  1%] Building CUDA object libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o
/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(18): error: qualified name is not allowed

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(82): error: this declaration has no storage class or type specifier

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(82): error: expected a ";"

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: this declaration has no storage class or type specifier

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: "numeric_cuda" has already been declared in the current scope

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/include/private/numeric_cuda.h(83): error: expected a ";"

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(70): error: name must be a namespace name

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(71): error: name must be a namespace name

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(71): error: expected a ";"

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: name followed by "::" must be a class or namespace name

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: expected an identifier

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(89): error: expected a ";"

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: name followed by "::" must be a class or namespace name

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: expected an identifier

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(90): error: expected a ";"

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(91): error: name followed by "::" must be a class or namespace name

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(91): error: expected an identifier

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(91): error: expected a ";"

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(94): error: identifier "R" is undefined

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(95): error: identifier "Ki" is undefined

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(97): error: identifier "t" is undefined

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: name followed by "::" must be a class or namespace name

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: expected an identifier

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(100): error: expected a ";"
[... 66 lines cut ...]

/home/xiechen/tandem/tandem/libdr/cuda_coarse_tracker/src/cuda_coarse_tracker_private.cu(134): warning: variable "ptT2_s" was declared but never referenced

[... cut ...]
libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/build.make:89: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o' failed
make[2]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/src/cuda_coarse_tracker_private.cu.o] Error 1
CMakeFiles/Makefile2:274: recipe for target 'libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/all' failed
make[1]: *** [libdr/cuda_coarse_tracker/CMakeFiles/cuda_coarse_tracker.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2
`

wtpro, 2022-03-11:
i got this exact error. It is not due to gcc/g++ version. I have tried 7 and 8 they both did not work.

Xie-HIT, 2022-03-11:
You are right, since my gcc version is 7.5.0. By the way, what is your CUDA version? My CUDA version is 10.2(also NVCC version),  will it be related to this?

wtpro, 2022-03-11:
Actually yes. I am using CUDA 10.2 which might be the problem here.

Xie-HIT, 2022-03-12:
Yes, it is a NVCC version problem. I finally solve this question by installing CUDA11.1 and CuDNN8.0.4 in my computer. Here is the bolg I referenced: [https://towardsdatascience.com/installing-multiple-cuda-cudnn-versions-in-ubuntu-fcb6aa5194e2](url)
After setting $PATH, and adding `-DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-11.1` in my CMake options, I successfully built target tandem_dataset : )

closed 2022-03-12 by Xie-HIT

three-nine, 2022-05-07:
https://github.com/tum-vision/tandem/issues/26
Hi,this problem is disturbing me, can you help me solve this problem?I have added -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-11.1 in my CMake options.