tum-vision/tandem #21: TypeError when train CVA-MVSNet

YuhsiHu, 2022-03-24:
Thank you for sharing this great paper and excellent code.
I followed the README file in CVA-MVSNet. When I tried to train the model, there was a TypeError. Could you please help me? Thank you for your time.

> GPU available: True, used: True
INFO:lightning:GPU available: True, used: True
No environment variable for node rank defined. Set as 0.
WARNING:lightning:No environment variable for node rank defined. Set as 0.
CUDA_VISIBLE_DEVICES: [0]
INFO:lightning:CUDA_VISIBLE_DEVICES: [0]
Validation sanity check: 0it [00:00, ?it/s]/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torch/functional.py:568: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  ../aten/src/ATen/native/TensorShape.cpp:2228.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Epoch 1:   0%|                                                                                                                                                                                                      | 0/2714 [00:00<?, ?it/s]Traceback (most recent call last):
  File "train.py", line 126, in <module>
    main(*parse_args(parser))
  File "train.py", line 122, in main
    trainer.fit(model)
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/trainer.py", line 859, in fit
    self.single_gpu_train(model)
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/distrib_parts.py", line 503, in single_gpu_train
    self.run_pretrain_routine(model)
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/trainer.py", line 1015, in run_pretrain_routine
    self.train()
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/training_loop.py", line 347, in train
    self.run_training_epoch()
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/training_loop.py", line 406, in run_training_epoch
    enumerate(_with_is_last(train_dataloader)), "get_train_batch"
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/profiler/profilers.py", line 64, in profile_iterable
    value = next(iterator)
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/pytorch_lightning/trainer/training_loop.py", line 801, in _with_is_last
    last = next(it)
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 530, in __next__
    data = self._next_data()
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 1224, in _next_data
    return self._process_data(data)
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 1250, in _process_data
    data.reraise()
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torch/_utils.py", line 457, in reraise
    raise exception
**TypeError: Caught TypeError in DataLoader worker process 0.**
Original Traceback (most recent call last):
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/hyx/tandem/cva_mvsnet/models/datasets.py", line 583, in __getitem__
    item = self.dataset.__getitem__(idx)
  File "/home/hyx/tandem/cva_mvsnet/models/datasets.py", line 562, in __getitem__
    data = self.transform(data)
  File "/home/hyx/tandem/cva_mvsnet/models/datasets.py", line 97, in preprocess
    img_aug = color_trans(img_ori)
  File "/home/hyx/anaconda3/envs/tandem/lib/python3.7/site-packages/torchvision/transforms/transforms.py", line 95, in __call__
    img = t(img)
**TypeError: 'tuple' object is not callable**

YuhsiHu, 2022-06-08:
No, I didn't.

wyt1004, 2022-07-18:
I guess this may be caused by different pytorch versions.
The transforms.ColorJitter.get_params function in datasets.py returns a tuple, and the tuple cannot be called as a function in the transforms.Compose function.
I tried a working method, modifying datasets.py:

fn_idx, brightness_factor, contrast_factor, saturation_factor, hue_factor = transforms.ColorJitter.get_params(
            brightness=brightness,
            contrast=contrast,
            saturation=saturation,
            hue=hue
        )
color_trans = transforms.Compose([
        transforms.ToPILImage(),
        transforms.ColorJitter(brightness=brightness_factor,
            contrast=contrast_factor,
            saturation=saturation_factor),
    ])
Hope this helps you.