tum-vision/tandem #22: RUNNING --- results/tracking/dense/euroc/V1_01_easy/0 [W BinaryOps.cpp:467] Warning: floor_divide is deprecated, and will be removed in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values. To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor'). (function operator())

zhlx007, 2022-03-29:
Dear @imoinuddin:
       I am writing for your help,When I enter and execute"bash scripts/tracking_euroc.bash",it display an warning ,and pass half hour,it exited,the warning is :
[W BinaryOps.cpp:467] Warning: floor_divide is deprecated, and will be removed in a future version of pytorch. It currently rounds toward 0 (like the 'trunc' function NOT 'floor'). This results in incorrect rounding for negative values.
To keep the current behavior, use torch.div(a, b, rounding_mode='trunc'), or for actual floor division, use torch.div(a, b, rounding_mode='floor'). (function operator())
      In the result folder,I can find the those files such as evaluate_ate.png ,mesh.obj,poses_dso.txt,keyframes_dso.txt  out.txt ,result.txt,please help me!!!!!!!

stihuangyuan, 2022-11-09:
> and

run without gui?

PatrickSDUST, 2023-08-03:
hi, have you solved this problem? I meet it too