tum-vision/tandem #25: Linking error to pangolin

loveforeverLi, 2022-04-15:
Hi,

Thanks for making this great work open source.

My compilation env: Ubuntu18.04 + cmake 3.22.1 + cuda 11.3 + Pangolin (0.6)

The compilation was done, but when linking to pangolin, the following error was reported:
`
[100%] Linking CXX executable bin/tandem_dataset
CMakeFiles/tandem_dataset.dir/src/main_tandem_pangolin.cpp.o: In function `main::{lambda()#1}::operator()() const':
/home/sky/project/tandem_workspace/tandem/tandem/src/main_tandem_pangolin.cpp:299: undefined reference to `DrFusion::SaveMeshToFile(std::string const&, float*, float*)'
lib/libtandem.a(PangolinDSOViewer.cpp.o): In function `dso::IOWrap::PangolinDSOViewer::run()':
/home/sky/project/tandem_workspace/tandem/tandem/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp:111: undefined reference to `pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)'
/home/sky/project/tandem_workspace/tandem/tandem/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp:132: undefined reference to `pangolin::Display(std::string const&)'
/home/sky/project/tandem_workspace/tandem/tandem/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp:140: undefined reference to `pangolin::Display(std::string const&)'
/home/sky/project/tandem_workspace/tandem/tandem/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp:143: undefined reference to `pangolin::Display(std::string const&)'
/home/sky/project/tandem_workspace/tandem/tandem/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp:185: undefined reference to `pangolin::CreatePanel(std::string const&)'
lib/libtandem.a(PangolinDSOViewer.cpp.o): In function `void pangolin::InitialiseNewVarMeta<double>(pangolin::VarValue<double>&, std::string const&, double, double, int, bool)':
/usr/local/include/pangolin/var/var.h:66: undefined reference to `pangolin::Split(std::string const&, char)'
lib/libtandem.a(PangolinDSOViewer.cpp.o): In function `void pangolin::InitialiseNewVarMeta<bool>(pangolin::VarValue<bool>&, std::string const&, double, double, int, bool)':
/usr/local/include/pangolin/var/var.h:66: undefined reference to `pangolin::Split(std::string const&, char)'
lib/libtandem.a(PangolinDSOViewer.cpp.o): In function `void pangolin::InitialiseNewVarMeta<int>(pangolin::VarValue<int>&, std::string const&, double, double, int, bool)':
/usr/local/include/pangolin/var/var.h:66: undefined reference to `pangolin::Split(std::string const&, char)'
collect2: error: ld returned 1 exit status
CMakeFiles/tandem_dataset.dir/build.make:176: recipe for target 'bin/tandem_dataset' failed
make[2]: *** [bin/tandem_dataset] Error 1
CMakeFiles/Makefile2:170: recipe for target 'CMakeFiles/tandem_dataset.dir/all' failed
make[1]: *** [CMakeFiles/tandem_dataset.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2 
`

I can compile dso successfully, so I compare the CMakeLists.txt between dso and tandem, but nothing was found. And I tried different version of pangolin, the above error reported. 

I have no idea about this error, any comments will be appreciated.

sunxirui310, 2022-05-19:
same error.lots of "undefined reference to pangolin::xxxxx"

frspring, 2022-11-03:
solved. The libtorch abi is not compatiable with pangolin. just reinstall another abi version of libtorch