tum-vision/tandem #26: Cudatoolkit couldn't find

Bobkk-k, 2022-04-28:
Hi, 

I installed CUDAToolkit 11.1.0 as per the nvidia official website. When I ran nvcc -V, the terminal displayed the text in the first 

image. I believe It means the installation is complete. However, when I ran 'cmake ..', an error happened(the second image).

Can anyone help me solve this problem?

Thank you!
[image]

[image]

three-nine, 2022-05-03:
Hi, have you solved this problem? I have the same problem too.

Bobkk-k, 2022-05-03:
Hi @three-nine,
Sorry to say I don't know how to solve it. If you find the method, please tell me. Thanks!

Bobkk-k, 2022-05-10:
Hi  @three-nine ,
I changed my cmake version and the error disappeared.

three-nine, 2022-05-10:
@Bobkk-k Thank you very much!

closed 2022-05-15 by Bobkk-k

wenmoqiu, 2022-11-12:
Hi @Bobkk-k ,
What is your cmake version? Mine is 3.16 but it didn't work