tum-vision/tandem #27: make error about Sophus

Bobkk-k, 2022-05-10:
make error about Sophus

[image]

When I fellow  the readme in the Sophus folder to make Sophus, something that looks likes make fail appeared.

[image]

Does anyone know how to solve it ?
Thanks!

Bobkk-k, 2022-05-15:
I changed the 'Sophus::SE3 <double>' in the error file to 'Sophus::SE3d' and the error disappeared

closed 2022-05-15 by Bobkk-k

jarvis-huang, 2022-10-20:
Correction => should be to change **`Sophus::SE3<double>`** to **`Sophus::SE3d`**. Its definition is [here](https://github.com/tum-vision/tandem/blob/master/tandem/thirdparty/Sophus/sophus/se3.hpp#L36).