tum-vision/tandem #29: compilation terminated. Related to sophus/se3.hpp when run make -j

kaejong2, 2022-05-25:
$ cmake .. 
Proceeded without errors.

[image]

From what I've checked, 
the CMakeLists.txt file has include_directories - ${PROJECT_SOURCE_DIR}/thirdparty/Sophus, so it seems to be sophus/se3.hpp with relative path, is that correct?

It doesn't seem to be finding the path, how can I fix this?

three-nine, 2022-06-26:
hi,have you solved this problem?I met the same problem

Wjt-shift, 2022-09-28:
hi,i also met that problem?have you solved this problem?thank you

kaejong2, 2022-09-28:
sorry. I'm late to check.
Here's how I solved it : Installing Sophus
https://github.com/tum-vision/tandem/tree/master/tandem/thirdparty/Sophus

And!  Check dependencies and reinstall 
cmake(version 3.24)
cuda 
ubuntu version
libeigen3-dev
etc...

I solved it in June and did something else, so I can't remember.

Wjt-shift, 2022-09-29:
> sorry. I'm late to check. Here's how I solved it : Installing Sophus https://github.com/tum-vision/tandem/tree/master/tandem/thirdparty/Sophus
>
> And! Check dependencies and reinstall cmake(version [...]

thanks for your reply!
i follow the step installing Sophus,but when i make the Sophus,i get the error like that:

/opt/cmake-3.24.2-linux-x86_64/bin/cmake -S/home/wjt/data/code/TANDEM/tandem/thirdparty/Sophus -B/home/wjt/data/code/TANDEM/tandem/thirdparty/Sophus/build --check-build-system CMakeFiles/Makefile.cmake 0
/opt/cmake-3.24.2-linux-x86_64/bin/cmake -E cmake_progress_start /home/wjt/data/code/TANDEM/tandem/thirdparty/Sophus/build/CMakeFiles /home/wjt/data/code/TANDEM/tandem/thirdparty/Sophus/build//CMakeFiles/progress.marks
make  -f CMakeFiles/Makefile2 all
make[1]: 进入目录“/home/wjt/data/code/TANDEM/tandem/thirdparty/Sophus/build”
make[1]: 对“all”not need anything。
make[1]: 离开目录“/home/wjt/data/code/TANDEM/tandem/thirdparty/Sophus/build”
/opt/cmake-3.24.2-linux-x86_64/bin/cmake -E cmake_progress_start /home/wjt/data/code/TANDEM/tandem/thirdparty/Sophus/build/CMakeFiles 0

i do not know what it means,i also git clone Sophus from github and install Sophus,i succesufully install Sophus,but i also met the problem before uninstalling Sophus.

jarvis-huang, 2022-10-20:
Go inside `thirdparty/Sophus` and install `sophus` (follow [README](https://github.com/tum-vision/tandem/blob/master/tandem/thirdparty/Sophus/README)). Make sure to **`sudo make install`** at the end so that sophus can be found by `cuda_coarse_tracker`.