tum-vision/tandem #35: Can run without any visual window

1974685090, 2022-07-02:
(https://user-images.githubusercontent.com/91367508/176985320-60ae95da-2216-4fe6-8f83-53faa045151d.png)

three-nine, 2022-07-03:
Hi,sorry to bother you.Can you tell me when you finished 'make -j' what you do next?Is it to execute 'build/bin/tandem_dataset' directly?Thank you very much!

1974685090, 2022-07-04:
Yes， run 'build/bin/tandem_dataset'directly

1974685090, 2022-07-04:
Yes， run 'build/bin/tandem_dataset'directly @three-nine

closed 2022-07-04 by 1974685090

three-nine, 2022-07-10:
@1974685090 Thanks!

three-nine, 2022-07-23:
Hi,  sorry to bother you again.   Have you used your owned dataset to train successfully?

zhangjd1029, 2023-05-07:
> Hi, sorry to bother you again. Have you used your owned dataset to train successfully?
Have you run your own dataset yet? I've been making segmentation fault