tum-vision/tandem #49: cmake error

wenbin-hhhhh, 2022-12-19:
编译cmake失败，报错找不到CUDA库，是由于cmake的问题吗？使用的是CUDA 11-1和CMAKE 3-16

[image]

pytholic, 2022-12-19:
@wenbin-hhhhh you can use docker-image [here] (https://github.com/tum-vision/tandem/pull/46).

w8128835, 2023-02-24:
> @wenbin-hhhhh you can use docker-image [here](https://github.com/tum-vision/tandem/pull/46).

@pytholic Hi,I installed the sophus in thirdparty folder ,but I met a problem like this picture when I use make -j10 to build the project.Do you know how to solve it ?
[image]

pytholic, 2023-03-03:
@w8128835 Hi, sorry for the late reply.
I had to did into ky old logs to find something. Is this (issue)[https://github.com/tum-vision/tandem/issues/29] similar to yours?

pytholic, 2023-03-03:
Also you can pull kmage and try from the link below.

https://hub.docker.com/repository/docker/pytholic/tumtandem/general

w8128835, 2023-03-03:
> @w8128835 Hi, sorry for the late reply. I had to did into ky old logs to find something. Is this [issue](https://github.com/tum-vision/tandem/issues/29) similar to yours?

Yes,I have already solved it.Thank you very much!

w8128835, 2023-03-03:
> Also you can pull kmage and try from the link below.
> 
> https://hub.docker.com/repository/docker/pytholic/tumtandem/general

@pytholic Thank for your work and reply, this is a awesome work! By the way, Is this docker you mentioned in ReadMe before? 
[image]

Can I use this docker enviroment to run all tandem and CVA-MVSNet?

pytholic, 2023-03-14:
@w8128835 Sorry for the late reply. I created the docker for my personal use, and then thought it might be useful for other people as well. I created PR but there hasn't been any response.

It's been a while since I worked with it, but if I remember correctly, it contains all the dependencies and should work well for tandem and MVSNet.

w8128835, 2023-03-14:
> 
 It's convenient for people to run tandem and MVSNet.Thank you very much!I will try it.

xfDing815, 2023-04-19:
@w8128835 Hi,I have the same problem like you,how did you solve it?
error: ‘Sophus::SE3’ {aka ‘Sophus::SE3Group<double>’} is not a template

nirmalsnair, 2023-07-16:
@pytholic Thank you very much for the docker image. As instructed, I have downloaded the built image using:
`docker pull pytholic/tumtandem:tumtandem`

But I don't understand what I should use in place of `work@xx.xx.xx.xx` in the next command `ssh -p 3751 work@xx.xx.xx.xx` as I am new to Linux (I am using Ubuntu 22).

pytholic, 2023-07-16:
@nirmalsnair Hi, can you take a look at instructions here:

https://github.com/pytholic/tandem/blob/master/docker/README.md

nirmalsnair, 2023-07-16:
@pytholic Yes, I am following those instructions only. I was referring to [this particular section](https://github.com/pytholic/tandem/blob/master/docker/README.md#run).

> Then in a new remote or local terminal, run the following command to connect the remote environment:
> ```
> ssh -p 3751 work@xx.xx.xx.xx
> \# default password: work123
> ```

pytholic, 2023-07-16:
@nirmalsnair I think instead of ssh,you can add -it flag to your docker run command. This will attach you terminal to the container.

pytholic, 2023-07-16:
@nirmalsnair or you can add -it flag to your docker run command.

nirmalsnair, 2023-07-16:
@pytholic Thanks. I ran `sudo docker run -it pytholic/tumtandem:tumtandem` and got this error message:

> nirmal@Razer:~/.ssh$ sudo docker run -it pytholic/tumtandem:tumtandem
> [sudo] password for nirmal: 
>  * Restarting OpenBSD Secure Shell server sshd                                                [ OK ] 
> Unable to init server: Could not connect: Connection refused
> Unable to init server: Could not connect: Connection refused
> You need to run terminator in an X environment. Make sure $DISPLAY is properly set

Sorry, I am new to Linux.

pytholic, 2023-07-16:
@nirmalsnair try adding -it flag after run in the following command.

docker run --rm \
  --name tandem \
  --ipc=host \
  --gpus all \
  --privileged \
  -p 3751:22 \         <======= please choose your own port
  -v /tmp/.X11-unix:/tmp/.X11-unix:rw \

pytholic, 2023-07-16:
@nirmalsnair you can modify run-docker-container.bash file and add -it flag there.

nirmalsnair, 2023-07-16:
@pytholic Ok, to do that I guess I cannot use the pre-built docker image. I tried building the docker image by cloning the repo and then:

```console
nirmal@Razer:~/tandem-master/docker$ chmod +x build-docker-image.bash
nirmal@Razer:~/tandem-master/docker$ sudo ./build-docker-image.bash
```

but I get an error in Step 42/59.

```console
Step 42/59 : RUN conda install -y python=3.7
 ---> Running in ccb0bfd9080f
Collecting package metadata (current_repodata.json): ...working... done
Solving environment: ...working... unsuccessful initial attempt using frozen solve. Retrying with flexible solve.
Solving environment: ...working... unsuccessful attempt using repodata from current_repodata.json, retrying with next repodata source.
Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... unsuccessful initial attempt using frozen solve. Retrying with flexible solve.
Solving environment: ...working... 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining conflict for pyopenssl urllib3 conda requests conda-libmamba-solver:   0%|          | 0/74 Examining conflict for pyopenssl conda pycparser brotlipy toolz libedit jsonpatch pysocks requests tqdm jsonpointer cryptography setuptools conda-content-trust readline wheel krb5 python boltons packaging urllib3 cffi six idna charset-normalizer pluggy ruamel.yaml conda-libmamba-solver pip zstandard libmambapy pycosat certifi conda-package-streaming conda-package-handling sqlite ncurses:   1%|▏       Examining conflict for pyopenssl conda pycparser brotlipy toolz libedit jsonpatch pysocks requests tqdm jsonpointer cryptography setuptools conda-content-trust readline wheel krb5 python boltons packaging urllib3 cffi six idna charset-normalizer pluggy ruamel.yaml conda-libmamba-solver pip zstandard libmambapy pycosat certifi conda-package-streaming conda-package-handling sqlite ncurses:   3%|▎       Examining conflict for pyopenssl conda pycparser libnghttp2 brotlipy toolz libssh2 libmamba jsonpatch pysocks requests openssl tqdm jsonpointer cryptography setuptools conda-content-trust libcurl krb5 wheel python boltons packaging urllib3 cffi six idna charset-normalizer pluggy ruamel.yaml conda-libmamba-solver pip zstandard libmambapy pycosat certifi conda-package-streaming conda-package-handling liExamining conflict for pyopenssl conda pycparser brotlipy toolz jsonpatch pysocks requests tqdm jsonpointer cryptography setuptools conda-content-trust tzdata wheel boltons packaging urllib3 cffi six idna charset-normalizer pluggy ruamel.yaml conda-libmamba-solver pip zstandard libmambapy pycosat certifi conda-package-streaming conda-package-handling:   4%|▍         | 3/74 [00:02<00:04, 14.67it/s]    Examining conflict for pyopenssl conda pycparser brotlipy toolz jsonpatch pysocks requests tqdm jsonpointer cryptography setuptools conda-content-trust tzdata wheel boltons packaging urllib3 cffi six idna charset-normalizer pluggy ruamel.yaml conda-libmamba-solver pip zstandard libmambapy pycosat certiExamining conflict for pyopenssl yaml-cpp sqlite reproc-cpp conda pycparser libnghttp2 brotlipy toolz fmt libssh2 zlib icu libedit libmamba jsonpatch pysocks requests openssl tqdm cryptography jsonpointer setuptools conda-content-trust readline pcre2 libcurl libev libuuid krb5 wheel python libgcc-ng boltons packaging urllib3 cffi six idna reproc c-ares charset-normalizer pluggy zstd ruamel.yaml conda-libmamba-solver lz4-c pip zstandard tk xz pycosat libmambapy libxml2 certifi libsolv conda-package-streaming conda-package-handling bzip2 libarchive libffi ncurses:   5%|▌         | 4/74 [00:03<00:51,  Examining conflict for pyopenssl yaml-cpp sqlite reproc-cpp conda pycparser libnghttp2 brotlipy toolz fmt libssh2 zlib icu libedit libmamba jsonpatch pysocks requests openssl tqdm cryptography jsonpointer setuptools conda-content-trust readline pcre2 libcurl libev libuuid krb5 wheel python libgcc-ng boltons packaging urllib3 cffi six idna reproc c-ares charset-normalizer pluggy zstd ruamel.yaml conda-libmamba-solver lz4-c pip zstandard tk xz pycosat libmambapy libxml2 certifi libsolv conda-package-streaming conda-package-handling bzip2 libarchive libffi ncurses:   7%|▋         | 5/74 [00:03<00:58,  Examining conflict for pyopenssl conda pycparser brotlipy toolz jsonpatch pysocks requests tqdm jsonpointer cryptography setuptools conda-content-trust wheel python boltons packaging urllib3 cffi six idna charset-normalizer pluggy ruamel.yaml conda-libmamba-solver pip zstandard libmambapy pycosat certifi conda-package-streaming conda-package-handling sqlite:   7%|▋         | 5/74 [00:04<00:58,  1.17it/s]                                                                                                                                                                                                       Examining conflict for pyopenssl conda pycparser brotlipy toolz jsonpatch pysocks requests tqdm jsonpointer cryptography setuptools conda-content-trust wheel python boltons packaging urllib3 cffi six idna charset-normalizer pluggy ruamel.yaml conda-libmamba-solver pip zstandard libmambapy pycosat certifi conda-package-streaming conda-package-handling sqlite:   8%|▊         | 6/74 [00:04<01:05,  1.04itExamining conflict for pyopenssl zstandard urllib3 cffi brotlipy conda-content-trust conda-package-streaming cryptography:   8%|▊         | 6/74 [00:06<01:05,  1.04it/s]                                  [... cut ...]  conflict for conda-package-handling libarchive: : 76it [00:24, 12.25it/s]                  failed                                                                             

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (pip):

  - python=3.7 -> pip -> setuptools
  - python=3.7 -> pip -> wheel
```

[Full install log.txt](https://github.com/tum-vision/tandem/files/12063781/Install.log.txt)

nirmalsnair, 2023-07-17:
@pytholic I tried the following command as seen in `run-docker-container.bash`:

```
sudo docker run --rm
--name tandem
-p 3751:22
-v /tmp/.X11-unix:/tmp/.X11-unix:rw
-v $HOME/.Xauthority:$docker/.Xauthority
-v $HOME/work:/home/work/projects
-v /mnt/Data/Datasets/dm-vio:/mnt/Data/Datasets/dm-vio
-e XAUTHORITY=$home_folder/.Xauthority
-e DISPLAY=$DISPLAY
-e QT_X11_NO_MITSHM=1
-it pytholic/tumtandem:tumtandem
```

and got the following error:

```console
nirmal@Razer:~/tandem-master/docker$ sudo docker run --rm --name tandem -p 3751:22 -v /tmp/.X11-unix:/tmp/.X11-unix:rw -v $HOME/.Xauthority:$docker/.Xauthority -v $HOME/work:/home/work/projects -v /mnt/Data/Datasets/dm-vio:/mnt/Data/Datasets/dm-vio -e XAUTHORITY=$home_folder/.Xauthority -e DISPLAY=$DISPLAY -e QT_X11_NO_MITSHM=1 -it pytholic/tumtandem:tumtandem
* Restarting OpenBSD Secure Shell server sshd                                                [ OK ] 
Authorization required, but no authorization protocol specified
Unable to init server: Could not connect: Connection refused
Authorization required, but no authorization protocol specified
Unable to init server: Could not connect: Connection refused
You need to run terminator in an X environment. Make sure $DISPLAY is properly set

```

However, after a few hours I was able to run the container using the same code.
[image]

But now, after a reboot, it has gone back to the previous error state. Any ideas on what the problem could be?