tum-vision/tandem #55: Convention of the Coordiante System / Representation of Poses

geoeo, 2023-07-20:
I couldn't find anything about this.

1) What do the poses in the pose file represent? Is it the pose of the camera in world coordinates w.r.t some other point(origin) also in world coordinates?
2) What Coordinate System should the poses be in? Is it Camera Coordinates i.e. +X - Right, +Y Down, +Z Into the image / Does it matter?

closed 2023-08-25 by geoeo

ShuhongLL, 2024-03-29:
Hello, did you figure out the coordinate system? I am facing the same problem. Thanks in advance

geoeo, 2024-04-02:
@ShuhongLL Poses should be in Camera Coordinates i.e +X - Right, +Y Down, +Z Into the image. 
Input pose file doesnt matter iirc the framework will tranform the poses such that the middle camera is the origin